openUC2/ImSwitch #16: Speed not implemented in stepper control

beniroquai, 2022-10-17:
Were you able to solve the issue with the z-axis speed control (i.e. currently moving 1000 um or 100 um at 10 um/s takes the same amount of time)?

kasasxav, 2022-10-17:
Hi @beniroquai,

I spotted a bug some time ago that I hadn't pushed yet, now I have commited it to [master](https://github.com/kasasxav/ImSwitch/commit/23078e0e907c68aa165383370a4b9fa28944ec64).
Could you check if this fixes your issues?

beniroquai, 2022-10-17:
Hey @kasasxav I checked the updates and pulled it into the branch. But I think this didn't resolve the issue. Something's weird. When hitting the "enter" button for "X", axis "Z" is selected - this doesn'T make sense, since everything is exactly the same way as e.g. the up/down for steps. 
I mean this region https://github.com/openUC2/ImSwitch/blob/cleanup/imswitch/imcontrol/view/widgets/PositionerWidget.py#L52

beniroquai, 2022-10-17:
Or perhaps the issue is caused by some wrong usage of the shared attributes? (no clue how it works anyway ;-) )
https://github.com/openUC2/ImSwitch/blob/cleanup/imswitch/imcontrol/controller/controllers/PositionerController.py#L25

closed 2023-01-15 by beniroquai