openUC2/ImSwitch #20: Could not build wheels for ImSwitchUC2

indev-aj, 2022-11-28:
When installing uc2 version on jetson nano, I received an error `ERROR: Could not build wheels for ImSwitchUC2, which is required to install pyproject.toml-based projects `

Installing from https://github.com/kasasxav/ImSwitch however works

Any ideas on this? Thank you.

beniroquai, 2023-01-15:
Hey @indev-aj sorry for my late reply. Indeed this is a known issue due to my inability to keep the version right. The github actions simply replaces this string `version-develop` to the release tag while uploading it to pipy. 
If you change https://github.com/openUC2/ImSwitch/blob/master/imswitch/__init__.py#L1 this to e.g. `v1.2.20` or something, it should work. 

@renerichter would you mind to check that on your end too? Did the installation go through?

renerichter, 2023-01-26:
@beniroquai I was not able to reproduce the problem on mac or windows, but I will try _Jetson nano_ soon. 
@indev-aj could you please briefly describe how you setup your Jetson? Please print the output of the commands:

```
cat /etc/os-release
arch
conda info
conda activate imswitch
conda list
pip list
```

indev-aj, 2023-01-26:
Sorry for the late reply. This is my detailed steps. There might be some difference in setup.cfg because it might have been updated. Although, if you meant installing from UC2, then i’m afraid I didn’t manage to install it, so I just use the other version.

```
1)	wget https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-Linux-aarch64.sh
2)	bash ./Miniforge3-Linux-aarch64.sh
3)	./anaconda3/bin/conda init
4)	conda create -n imswitch  python=3.9
5)	conda install pyqt=5.12.3 -y
6)	Git clone https://github.com/kasasxav/ImSwitch.git
7)	Cd ImSwitch
8)	Comment out PyQt, pywin32, Qsci in setup.cfg
9)	Pip install -r requirements-dev.txt
10)	Imswitch

```

You might also want to check out the instruction from [here](https://github.com/openUC2/ImSwitch#on-jetson-nano)

> @beniroquai I was not able to reproduce the problem on mac or windows, but I will try _Jetson nano_ soon. @indev-aj could you please briefly describe how you setup your Jetson? Please print the output of the commands:

beniroquai, 2023-12-03:
> also if some time is gone.
> try to install
> apt-get install -y qtbase5-dev
> bevor building imswitch

For the Jetson?