openUC2/ImSwitch #25: Blockly fails to load

renerichter, 2023-01-12:
**Bug description**
Imswitch not starting due to failing blockly. 

**How to reproduce**
Steps to reproduce the behavior:
1. install imswitch with all dependencies
2. git pull configs von git and put to ~/ImSwitchConfig
3. in `mamba activate imswitch` environment run `imswitch` command

**Expected behavior**
Assumed all modules (including blockly) to start properly.

**Output**

```
2023-01-12 16:11:02 ERROR [main] Failed to initialize module imblockly
2023-01-12 16:11:02 ERROR [main] Traceback (most recent call last):
  File "[...]ImSwitch/imswitch/__main__.py", line 58, in main
    view, controller = modulePkg.getMainViewAndController(
  File "[...]ImSwitch/imswitch/imblockly/__init__.py", line 55, in getMainViewAndController
    from .controller import ImScrMainController
  File "[...]ImSwitch/imswitch/imblockly/controller/__init__.py", line 1, in <module>
    from .ImScrMainController import ImScrMainController
  File "[...]ImSwitch/imswitch/imblockly/controller/ImScrMainController.py", line 5, in <module>
    from .ImScrMainViewController import ImScrMainViewController
  File "[...]ImSwitch/imswitch/imblockly/controller/ImScrMainViewController.py", line 2, in <module>
    from .EditorController import EditorController
  File "[...]ImSwitch/imswitch/imblockly/controller/EditorController.py", line 5, in <module>
    from imswitch.imblockly.view import guitools
  File "[...]ImSwitch/imswitch/imblockly/view/__init__.py", line 1, in <module>
    from .ImScrMainView import ImScrMainView
  File "[...]ImSwitch/imswitch/imblockly/view/ImScrMainView.py", line 4, in <module>
    from .BlocklyView import BlocklyView
  File "[...]ImSwitch/imswitch/imblockly/view/BlocklyView.py", line 3, in <module>
    from PyQt5.QtWebEngineWidgets import *
ImportError: QtWebEngineWidgets must be imported before a QCoreApplication instance is created

[1]    70295 segmentation fault  imswitch
/opt/homebrew/Caskroom/mambaforge/base/envs/imswitch/lib/python3.9/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 1 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
```

**Environment**
 - OS: osx-arm64
 - ImSwitch version: 1.2.11
 - How you downloaded ImSwitch: git

beniroquai, 2023-01-14:
Thanks @renerichter. This is indeed a known issue. I'll investigate. Most likely the Website is not being shipped through github correctly. I'll have a look.

beniroquai, 2023-01-14:
Seems I cannot even install the package under windwos..
https://stackoverflow.com/questions/51154871/python-3-7-0-no-module-named-pyqt5-qtwebenginewidgets

renerichter, 2023-01-26:
@beniroquai installation of the package (as it is shipped with PyQt5 as expected) is no problem. It seems that order of modules and packages that are imported is messed up. I will have a look.

beniroquai, 2023-01-26:
Yep. Napari is the issue I guess. Tried installing microeye for STORM reconstruction today which screwed up pydantic and pyqt5. Force-reinstalling resolved the issue in my case.

renerichter, 2023-02-01:
So a simple `pip install --upgrade --force-reinstall pyqt5` at the solved your problem? Maybe we should try this in the installation script then?

beniroquai, 2023-02-01:
How would that work? Sorry I have absolutely no clue what's possible with
setuptools at all ;-)

> Message ID: ***@***.***>
>

closed 2025-05-03 by beniroquai