openUC2/ImSwitch #29: First 100 frames are the same when acquiring1000 images in .hdf5 format

AZehrer, 2023-01-18:
Xaviers last response:
Alright! Thanks for reporting the bug btw. I think this manager I have not done it myself so I'm not sure, maybe it's from the UC2 repo? Can you take a look at getChunk and make sure that it's returning the new list of frames, in the camera manager? getChunk is the one that is called from the RecordingManager so that would be the one that is failing in my opinion

Probably more useful to post the issue in this repository.

AZehrer, 2023-01-18:
CMOS Alvium camera was used.

beniroquai, 2023-01-19:
you need to tag so that I can see it :D 

do @beniroquai ^^

AZehrer, 2023-01-25:
@beniroquai

beniroquai, 2023-03-27:
Thanks @AZehrer - sorry for taking soo long. Is this still an issue? 
I suspect it's due to the lack of the clear-buffer function in the vimba camera driver:
https://github.com/openUC2/ImSwitch/blob/082bcb34f0b7b2b97bf585abc9a08cf69d32c7fc/imswitch/imcontrol/model/interfaces/avcamera.py#L144

I remember that the `RecordingManager` had this function of "flushBuffer" before starting an acquisition series. @kasasxav was that true? In the Daheng camera I implemented it 

https://github.com/openUC2/ImSwitch/blob/082bcb34f0b7b2b97bf585abc9a08cf69d32c7fc/imswitch/imcontrol/model/interfaces/gxipycamera.py#L196

beniroquai, 2023-03-27:
Added it to the latest commit c6a98805749593ca1e9134b9893da47ff7bc201d

closed 2025-05-03 by beniroquai