openUC2/ImSwitch #32: Installing non-"piped" packages through pip install?

beniroquai, 2023-01-26:
When installing this repo from source, a few dependencies are missing. How can we ensure they are getting installed properly (asking for a friend :D @AZehrer).. 

One example:
https://github.com/openUC2/ImSwitch/blob/master/setup.cfg#L64

CC: @renerichter @kasasxav

kasasxav, 2023-01-29:
I don't really know the answer to this one sorry :L I would say either doing it manually after (https://stackoverflow.com/questions/7813030/how-can-i-have-linked-dependencies-in-a-git-repo) or maybe doing the .whl and then adding the file in the repo. Not ideal though and it will be a bit messy, maybe someone else has a better approach?

beniroquai, 2023-01-29:
That's a good start. I have tried several formulations in the `setup.cfg`, but it seems it's an OS/version-dependent issue. That makes it difficult ;-) 

@renerichter any thoughts on this? :)

renerichter, 2023-02-01:
@beniroquai you mean that commented packages exist that shall rather be uncommented? Or do you mean that when uncommenting these packages in setup.cfg installation breaks? How about adding e.g. a conda installation recipe which the user could easily invoke afterwards?
Finally, a batch-script could be created which does an auto-installation, but this is tedious to maintain for different OS and versions of many toolboxes.

beniroquai, 2023-02-01:
What I mean is this (already outcommented) line: https://github.com/openUC2/ImSwitch/blob/master/setup.cfg#L61
In best case, we have this in the setup.cfg. It pulls packages from github/gitlab and installs it similar to pypi packages. Makes sense?
Agree, bash scripts are cumbersome..

closed 2023-09-25 by beniroquai