openUC2/ImSwitch #38: PositionerManager vs. ESP32StageManager class conflict

mgstingl, 2023-02-24:
Hi @beniroquai !

The [ESP32StageManager ](https://github.com/openUC2/ImSwitch/blob/d67f9d975b04e931fa9e33dfa1e5ec1e2698a4ff/imswitch/imcontrol/model/managers/positioners/ESP32StageManager.py) doesn't use the base definition (doesn't respect the abstract methods) of the [PositionerManager](https://github.com/openUC2/ImSwitch/blob/d67f9d975b04e931fa9e33dfa1e5ec1e2698a4ff/imswitch/imcontrol/model/managers/positioners/PositionerManager.py), even when it inherits from it. 

1. The [_move_ method](https://github.com/openUC2/ImSwitch/blob/d67f9d975b04e931fa9e33dfa1e5ec1e2698a4ff/imswitch/imcontrol/model/managers/positioners/ESP32StageManager.py#L163) in the ESP32StageManager switches _dist_ for _value_, I guess to include the _is_absolute_ option, but conflicting with the abstract method.

2. In general, the ESP32Stage interface could include another abstraction layer that integrates the [motor and home modules](https://github.com/openUC2/ImSwitch/blob/d67f9d975b04e931fa9e33dfa1e5ec1e2698a4ff/imswitch/imcontrol/model/managers/positioners/ESP32StageManager.py#L138), in order to integrate it in the manager:
```
  self._motor = self._rs232manager._esp32.motor
  self._homeModule = self._rs232manager._esp32.home
```
These 2 lines would be simplified with the following line, which includes the business logic of the complete XYZ stage:
```
self._stage = self._rs232manager._esp32.stage
```
This would follow the convention of the Manager class instantiating a low-level interface, like for example in this [LantzLaserManager](https://github.com/openUC2/ImSwitch/blob/d67f9d975b04e931fa9e33dfa1e5ec1e2698a4ff/imswitch/imcontrol/model/managers/lasers/LantzLaserManager.py), where the Manager instantiates a self._laser:
```
class LantzLaserManager(LaserManager):
    def __init__(self, laserInfo, name, isBinary, valueUnits, valueDecimals, driver,
                 **_lowLevelManagers):
        self.__logger = initLogger(self, instanceName=name)
        ports = laserInfo.managerProperties['digitalPorts']
        # Init laser
        self._laser = LantzLaser(driver, ports)
```

Kind regards!

beniroquai, 2023-02-24:
Hey @mgstingl thanks for reporting that! I fully agree. 
What would you suggest in order to improve the structure? :)

mgstingl, 2023-03-03:
Hi @beniroquai !

I will try to provide some improvements while working on the scan script that can use a deck configuration and is independent of the stage used. When I get it working, I will push it to Labmaite's branch and comment it.

beniroquai, 2023-03-03:
In best case you can also file a PR @mgstingl  ;-)

beniroquai, 2023-09-25:
@mgstingl did you ever solve this issue? ;)

closed 2025-05-03 by beniroquai