openUC2/ImSwitch #47: Provide way to upload sim patterns to user folder

ranranking, 2023-03-10:
@beniroquai

beniroquai, 2023-03-13:
@ranranking What would be a good strategy? Just place files inside the user folder that are precomputed ?
What are the main parameters? Wavelength and effective pixelsize? 
We could have a dropdown menu that enables the selection of available experiments (e.g. 488,635,1.2NA,63x) or something?

ranranking, 2023-03-15:
i think upload patterns are a bit too much effort now and also a bit complicated. We need to match the wavelength with grating constant also the grating direction with the fourier mask. Maybe we keep it first like now and try to do it later

beniroquai, 2023-04-17:
@ranranking has this issue been resolved already? you can copy/paste the grating bmps into the folder right?

closed 2025-01-11 by beniroquai