openUC2/ImSwitch #52: ImSwitch hangs when recording

dboonz, 2023-03-24:
When I hit 'record' for a photometrics camera (PVCAM manager), imswitch hangs and dies eventually. 

Eventually it crashes with this error:

```python
2023-03-24 09:53:50 ERROR [PhotometricsManager -> WidefieldCamera] Traceback (most recent call last):
  File "C:\Users\dirk\PycharmProjects\imswitch_hackathon\ImSwitch\imswitch\imcontrol\model\managers\detectors\DetectorManager.py", line 126, in updateLatestFrame
    self.__image = self.getLatestFrame()
  File "C:\Users\dirk\PycharmProjects\imswitch_hackathon\ImSwitch\imswitch\imcontrol\model\managers\detectors\PhotometricsManager.py", line 68, in getLatestFrame
    return np.array(self._camera.poll_latest_frame()[0]['pixel_data'])
numpy.core._exceptions._ArrayMemoryError: Unable to allocate 19.5 MiB for an array with shape (3200, 3200) and data type uint16
```

This happens when saving as an 
- HDF5 file
- tiff

I didn't check the other ones yet.

I also get the same problem when I run the example_record.py script. It sounds like imswitch is not stopping.

beniroquai, 2023-06-09:
@jacopoabramo I guess you fixed that one?

jacopoabramo, 2023-06-09:
yep, it's fixed. it was a typo on the function

closed 2023-09-25 by beniroquai