openUC2/ImSwitch #62: LED Matrix is crashing when entering that in ImSwitch

beniroquai, 2024-03-19:
```py
mPattern = list(({"r":0, "g":1, "b":0},
			{"r":0, "g":0, "b":0}))
api.imcontrol.setSpecial(mPattern, 255, False)
```

@beniroquai 

```py
ttps_server: Server listening on port 80
INFO:imswitch:[ScriptExecutor] Started script
DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"led_array": [{"id": 0, "r": 0, "g": 0, "b": 0}], "LEDArrMode": 1}, "qid": 1}

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"r": 0, "g": 255, "b": 0}, {"r": 0, "g": 0, "b": 0}]}, "qid": 2}

INFO:imswitch:[ScriptExecutor] Finished script
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:++
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:++
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:{
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:{
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"qid": 1,
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"qid":    1,
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"success":     1
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"success":        1
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:}
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:}
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:--
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:--
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d45d0  PS      : 0x00060330  A0      : 0x800d56f5  A1      : 0x3fff2fa0
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d45d0  PS      : 0x00060330  A0      : 0x800d56f5  A1      : 0x3fff2fa0
2024-03-19 13:39:41 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd0388  A3      : 0x3ffbafb0  A4      : 0xfffffff7  A5      : 0x00000007
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd0388  A3      : 0x3ffbafb0  A4      : 0xfffffff7  A5      : 0x00000007
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x00000002  A7      : 0x40404040  A8      : 0x800d45cd  A9      : 0x3fff2f80
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x00000002  A7      : 0x40404040  A8      : 0x800d45cd  A9      : 0x3fff2f80
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f4453d2  A12     : 0x3f403f91  A13     : 0x00000000
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f4453d2  A12     : 0x3f403f91  A13     : 0x00000000
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x3ffc4438  A15     : 0x00000082  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x3ffc4438  A15     : 0x00000082  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:EXCVADDR: 0x00000014  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x801243ea
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:EXCVADDR: 0x00000014  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x801243ea
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Backtrace: 0x400d45cd:0x3fff2fa0 0x400d56f2:0x3fff2fd0 0x400d5a42:0x3fff2ff0 0x400d2086:0x3fff3020
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Backtrace: 0x400d45cd:0x3fff2fa0 0x400d56f2:0x3fff2fd0 0x400d5a42:0x3fff2ff0 0x400d2086:0x3fff3020
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:ELF file SHA256: 048e86987eebb6c5
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:ELF file SHA256: 048e86987eebb6c5
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Rebooting...
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Rebooting...
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (740) HIDGamePad: setup
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (740) HIDGamePad: setup
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (741) HIDGamePad: setting hid gap, mode:3
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (741) HIDGamePad: setting hid gap, mode:3
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (741) BTDM_INIT: BT controller compile version [8020d24]
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (741) BTDM_INIT: BT controller compile version [8020d24]
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (742) system_api: Base MAC address is not set
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (742) system_api: Base MAC address is not set
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (742) system_api: read default base MAC address from EFUSE
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (742) system_api: read default base MAC address from EFUSE
2024-03-19 13:39:42 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (746) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (746) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
2024-03-19 13:39:43 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1215) HIDGamePad: EVENT: 5
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1215) HIDGamePad: EVENT: 5
2024-03-19 13:39:43 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1216) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1216) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
2024-03-19 13:39:43 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1216) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1216) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:0 
2024-03-19 13:39:43 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1218) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1218) gpio: GPIO[17]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[... 198 lines cut ...]
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1771) wifi_init: udp mbox: 6
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1771) wifi_init: udp mbox: 6
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1772) wifi_init: tcp mbox: 6
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1772) wifi_init: tcp mbox: 6
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1773) wifi_init: tcp tx win: 5744
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1773) wifi_init: tcp tx win: 5744
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1774) wifi_init: tcp rx win: 5744
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1774) wifi_init: tcp rx win: 5744
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1775) wifi_init: tcp mss: 1440
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1775) wifi_init: tcp mss: 1440
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1783) wifi:mode : sta (a0:b7:65:50:06:d0) + softAP (a0:b7:65:50:06:d1)
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1783) wifi:mode : sta (a0:b7:65:50:06:d0) + softAP (a0:b7:65:50:06:d1)
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1783) wifi:enable tsf
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1783) wifi:enable tsf
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1784) wifi:Total power save buffer number: 16
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1784) wifi:Total power save buffer number: 16
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1784) wifi:Init max length of beacon: 752/752
[... cut ...]
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1785) wifi:Init max length of beacon: 752/752
2024-03-19 13:40:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1787) esp_https_server: Starting server
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1787) esp_https_server: Starting server
2024-03-19 13:40:08 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (1790) esp_https_server: Server listening on port 80
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (1790) esp_https_server: Server listening on port 80
```

beniroquai, 2024-03-21:
Some more investigation:

the following string:
```json
{
	"task": "/ledarr_act",
	"led": {
	  "LEDArrMode": 0,
	  "led_array": [
		{ "id": 0, "r": 0, "g": 255, "b": 0 },
		{ "id": 1, "r": 0, "g": 255, "b": 0 },
		{ "id": 2, "r": 0, "g": 255, "b": 0 },
		{ "id": 3, "r": 0, "g": 255, "b": 0 },
		{ "id": 4, "r": 0, "g": 255, "b": 0 },
		{ "id": 5, "r": 0, "g": 255, "b": 0 },
		{ "id": 6, "r": 0, "g": 255, "b": 0 },
		{ "id": 7, "r": 0, "g": 255, "b": 0 },
		{ "id": 8, "r": 0, "g": 0, "b": 0 },
		{ "id": 9, "r": 0, "g": 0, "b": 0 },
		{ "id": 10, "r": 0, "g": 0, "b": 0 },
		{ "id": 11, "r": 0, "g": 0, "b": 0 },
		{ "id": 12, "r": 0, "g": 0, "b": 0 },
		{ "id": 13, "r": 0, "g": 0, "b": 0 },
		{ "id": 14, "r": 0, "g": 0, "b": 0 },
		{ "id": 15, "r": 0, "g": 0, "b": 0 }
	  ]
	},
	"qid": 47
  }
```

gave the following error:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x400d574c  PS      : 0x00060c30  A0      : 0x800d696d  A1      : 0x3fff5b10
A2      : 0x3ffbd844  A3      : 0x3ffbb900  A4      : 0x00000007  A5      : 0x00000000
A6      : 0x0000002f  A7      : 0x000000ff  A8      : 0x800d5749  A9      : 0x3fff5af0
A10     : 0x00000000  A11     : 0x3f445ee2  A12     : 0x00000000  A13     : 0x0000ff00
A14     : 0x00000000  A15     : 0x0000018f  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000014  LBEG    : 0x40084201  LEND    : 0x40084209  LCOUNT  : 0x00000027
Backtrace: 0x400d5749:0x3fff5b10 0x400d696a:0x3fff5b40 0x400d6cba:0x3fff5b60 0x400d21ae:0x3fff5b90
ELF file SHA256: 60cc6a73c0b6f0a7

PC: 0x400d574c: LedController::act(cJSON*) at main/src/led/LedController.cpp line 64
EXCVADDR: 0x00000014

Decoding stack results
0x400d5749: LedController::act(cJSON*) at main/src/led/LedController.cpp line 64
0x400d696a: SerialProcess::jsonProcessor(char*, cJSON*) at main/src/serial/SerialProcess.cpp line 321
0x400d6cba: SerialProcess::loop() at main/src/serial/SerialProcess.cpp line 64
0x400d21ae: looper(void*) at main/main.cpp line 20
```

and suggests that the parsed values are not correct:
```cpp
  cJSON_ArrayForEach(arri, ledarr)
  {
	  set_led_RGB(
				  cJSON_GetObjectItemCaseSensitive(arri,keyid)->valueint,
				  cJSON_GetObjectItemCaseSensitive(arri,keyRed)->valueint,
				  cJSON_GetObjectItemCaseSensitive(arri,keyGreen)->valueint,
				  cJSON_GetObjectItemCaseSensitive(arri,keyBlue)->valueint);

  }
```
as @KillerInk suggested 

I changed multiple things:
- checked values in the firmware if None/available 
- increased write-timeout in python 

Things to do:
- update ImSwitch, UC2-REST and Firmware (from youseetoo.github.io)

I thin with this it should not crash anymore :)

closed 2024-03-21 by beniroquai

AaronRiSa, 2024-03-26:
Hi! @beniroquai 
I am trying to use the script module in ImSwitch to program my LED array with different illumination patterns. I used two different ESP32 units and my code is not working properly. Does this mean ImSwitch/UC2-REST and the Firmware have not been updated? Because this is still not working. I run the script and the LED pattern is not shown properly (some LEDs are still off) then the ESP32 reboots and the LED array is all off.

This is my current code and outputs:

**Input in script window:**

mPattern = list(({"id":1,"r":0, "g":1, "b":0},
{"id":2, "r":0, "g":1, "b":0},
{"id":3, "r":0, "g":1, "b":0},
{"id":4, "r":0, "g":1, "b":0},
{"id":5, "r":0, "g":1, "b":0},
{"id":6, "r":0, "g":1, "b":0},
{"id":7, "r":0, "g":1, "b":0},
{"id":8, "r":0, "g":1, "b":0},
{"id":9, "r":0, "g":1, "b":0},
{"id":10, "r":0, "g":1, "b":0},
{"id":11, "r":0, "g":1, "b":0},
{"id":12, "r":0, "g":1, "b":0},
{"id":13, "r":0, "g":1, "b":0},
{"id":14, "r":0, "g":1, "b":0},
{"id":15, "r":0, "g":1, "b":0},
{"id":0, "r":0, "g":1, "b":0}))
api.imcontrol.setSpecial(mPattern, 255, False)

**Output in script window:**

2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0}, {"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 14}

2024-03-26 09:22:04 INFO [ScriptExecutor] Finished script

**Output in VCode:**

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0},
{"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 14}

INFO:imswitch:[ScriptExecutor] Finished script
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:++
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:++
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:{
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:{
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"qid": 13,
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"qid":    13,
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"success":     1
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"success":        1
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:}
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:}
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:--
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:--
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d574c  PS      : 0x00060630  A0      : 0x800d696d  A1      : 0x3fff5cd0
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d574c  PS      : 0x00060630  A0      : 0x800d696d  A1      : 0x3fff5cd0
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd03f8  A3      : 0x3ffbb584  A4      : 0x00000008  A5      : 0x00000000
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd03f8  A3      : 0x3ffbb584  A4      : 0x00000008  A5      : 0x00000000
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x0000000e  A7      : 0x000000ff  A8      : 0x800d5749  A9      : 0x3fff5cb0
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x0000000e  A7      : 0x000000ff  A8      : 0x800d5749  A9      : 0x3fff5cb0
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f445ee2  A12     : 0x00000000  A13     : 0x0000ff00
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f445ee2  A12     : 0x00000000  A13     : 0x0000ff00
2024-03-26 09:22:04 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x00000000  A15     : 0x0000019e  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x00000000  A15     : 0x0000019e  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c
[... cut ...]
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2451) wifi:Init max length of beacon: 752/752
2024-03-26 09:22:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (2452) esp_https_server: Starting server
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2452) esp_https_server: Starting server
2024-03-26 09:22:07 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (2456) esp_https_server: Server listening on port 80
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2456) esp_https_server: Server listening on port 80

Thank you! I will stay attentive :)

beniroquai, 2024-03-26:
The error that comes out looks similar to the one we had before. 
I updated the LED Matrix code in the firmware here:
https://github.com/youseetoo/uc2-esp32/blob/main/main/src/led/LedController.cpp#L64

The issue was I think a mix of timeout and incomplete JSON which was misinterpreted by the ESP32. 

Would you mind to do two things from this webpage: https://youseetoo.github.io/indexWebSerialTest.html

1. Connect to the ESP 
2. type `{"task":"/state_get"}` (and print the output - if the compile date is older than 10.3. you would need to update)
3. type `{"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0}, {"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 14}` and paste what happens 

It looks like a proper Json string for the LED Matrix, so there must be something wrong from the Python side (e.g. timeout, interrupting the write process or something). 

Let me know what's happening there. Sänk ju!

beniroquai, 2024-03-26:
what's the compiled date? 

I guess you can update UC2-REST as well (most likely from github desktop) or do `pip install https://github.com/openUC2/UC2-REST/archive/refs/heads/master.zip`

The relevant line might be this: 
https://github.com/openUC2/UC2-REST/blob/master/uc2rest/mserial.py#L192

The correct information is supposed to be sent, but it most likely doesn't reach the ESP in the proper format - the reboot suggests that.

AaronRiSa, 2024-03-26:
The compiled date was 20.03 which I think is fine. Then I tried updating the UC2-REST. I double checked the relevant lines and it is still not working. It works using the LED Widget all the time, but once I use the script window it shows the pattern and immediatly reboots.

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"led_array": [{"id": 0, "r": 0, "g": 0, "b": 0}], "LEDArrMode": 1}, "qid": 53}

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0}, 
{"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 54}

INFO:imswitch:[ScriptExecutor] Finished script
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:++
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:++
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:{
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:{
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"qid": 53,
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"qid":    53,
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"success":     1
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"success":        1
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:}
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:}
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:--
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:--
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Core  1 register dump:
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d574c  PS      : 0x00060630  A0      : 0x800d696d  A1      : 0x3fff5d80
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:PC      : 0x400d574c  PS      : 0x00060630  A0      : 0x800d696d  A1      : 0x3fff5d80
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd03f8  A3      : 0x3ffbb8ec  A4      : 0x00000008  A5      : 0x00000000
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A2      : 0x3ffd03f8  A3      : 0x3ffbb8ec  A4      : 0x00000008  A5      : 0x00000000
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x00000036  A7      : 0x000000ff  A8      : 0x800d5749  A9      : 0x3fff5d60
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A6      : 0x00000036  A7      : 0x000000ff  A8      : 0x800d5749  A9      : 0x3fff5d60
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f445ee2  A12     : 0x00000000  A13     : 0x0000ff00
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A10     : 0x00000000  A11     : 0x3f445ee2  A12     : 0x00000000  A13     : 0x0000ff00
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x00000000  A15     : 0x0000019e  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:A14     : 0x00000000  A15     : 0x0000019e  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:EXCVADDR: 0x00000014  LBEG    : 0x40084201  LEND    : 0x40084209  LCOUNT  : 0x00000027
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:EXCVADDR: 0x00000014  LBEG    : 0x40084201  LEND    : 0x40084209  LCOUNT  : 0x00000027
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Backtrace: 0x400d5749:0x3fff5d80 0x400d696a:0x3fff5db0 0x400d6cba:0x3fff5dd0 0x400d21ae:0x3fff5e00
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Backtrace: 0x400d5749:0x3fff5d80 0x400d696a:0x3fff5db0 0x400d6cba:0x3fff5dd0 0x400d21ae:0x3fff5e00
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:ELF file SHA256: 60cc6a73c0b6f0a7
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:ELF file SHA256: 60cc6a73c0b6f0a7
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:Rebooting...
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:Rebooting...
2024-03-26 13:21:12 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (757) HIDGamePad: setting hid gap, mode:3
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (757) HIDGamePad: setting hid gap, mode:3
2024-03-26 13:21:12 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (757) BTDM_INIT: BT controller compile version [8020d24]
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (757) BTDM_INIT: BT controller compile version [8020d24]
2024-03-26 13:21:12 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (758) system_api: Base MAC address is not set
[... cut ...]
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2526) wifi:Init max length of beacon: 752/752
2024-03-26 13:21:14 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (2527) esp_https_server: Starting server
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2527) esp_https_server: Starting server
2024-03-26 13:21:14 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:I (2531) esp_https_server: Server listening on port 80
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:I (2531) esp_https_server: Server listening on port 80

beniroquai, 2024-03-26:
The output suggests that there is a return message from the i"all leds on" from `qid=53`, but ImSwitch already sends another message. 

```py
DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"led_array": [{"id": 0, "r": 0, "g": 0, "b": 0}], "LEDArrMode": 1}, "qid": 53}

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendMessage]: {"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0},
{"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 54}

INFO:imswitch:[ScriptExecutor] Finished script
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:++
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:++
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:{
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:{
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"qid": 53,
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"qid": 53,
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:"success": 1
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:"success": 1
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:}
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:}
2024-03-26 13:21:11 DEBUG [ESP32Manager -> ESP32] [ProcessLines]:--
DEBUG:imswitch:[ESP32Manager -> ESP32] [ProcessLines]:--
```

Where do you send the initial message 
`{"task": "/ledarr_act", "led": {"led_array": [{"id": 0, "r": 0, "g": 0, "b": 0}], "LEDArrMode": 1}, "qid": 53}`?
Is this also inside ImScript?

beniroquai, 2024-03-27:
Interesting. What happens if you hit individual LEDs from the LED Widget? I have no clue what's going on here. 
Maybe @KillerInk has some ideas? Why the ESP should reboot - it doesn't even give any location in code where it crashed (like the exception decoder..). Odd. 

One other thing we could try is the following. The interface (as long as Lasers are enabled in the configuration) has a "sendSerial" endpoint (like the one I have statically hosted here https://youseetoo.github.io/imswitch/api.html#/default/send_serial_LaserController_send_serial_get). 
So you can try the following code in the ImScripting:
```py
api.imcontrol.send_serial('{"task": "/ledarr_act", "led": {"LEDArrMode": 0, "led_array": [{"id": 1, "r": 0, "g": 255, "b": 0}, {"id": 2, "r": 0, "g": 255, "b": 0}, {"id": 3, "r": 0, "g": 255, "b": 0}, {"id": 4, "r": 0, "g": 255, "b": 0}, {"id": 5, "r": 0, "g": 255, "b": 0}, {"id": 6, "r": 0, "g": 255, "b": 0}, {"id": 7, "r": 0, "g": 255, "b": 0}, {"id": 8, "r": 0, "g": 255, "b": 0}, {"id": 9, "r": 0, "g": 255, "b": 0},
{"id": 10, "r": 0, "g": 255, "b": 0}, {"id": 11, "r": 0, "g": 255, "b": 0}, {"id": 12, "r": 0, "g": 255, "b": 0}, {"id": 13, "r": 0, "g": 255, "b": 0}, {"id": 14, "r": 0, "g": 255, "b": 0}, {"id": 15, "r": 0, "g": 255, "b": 0}, {"id": 0, "r": 0, "g": 255, "b": 0}]}, "qid": 54}')
```

KillerInk, 2024-03-27:
Looks like the json exceed the main task memory.
Pls try to send only one led and then increase its count till it crash. I expect it to crash after 5-6 LEDs inside the array.
Probably it's better to split the request into multiply request bevor increasing the tasksize.
So first request update led 1-4.
Second update 5-9....

KillerInk, 2024-03-27:
that problem is on py side.
can send here, per pio/cmd serial, the commands without a problem to the esp.

what seems to happen is that the serial gets blocked till it returns that "qid"(....) and that second command get never executed or added to the queue.

https://github.com/openUC2/UC2-REST/blob/ddf92725361fa51bcb082006852c44547a65b098/uc2rest/mserial.py#L324-L360

https://github.com/openUC2/UC2-REST/blob/ddf92725361fa51bcb082006852c44547a65b098/uc2rest/mserial.py#L349C9-L349C28

blocks the method till it have a response. but what to the second command happen while it blocks i dont know.
you have to wait till command 1 finished, and then you can execute the second one.

beniroquai, 2024-04-08:
Yup, agree. I wanted to implement a synchronous/asynchronous behavior, somehow. I wonder how async is doing it in javascript or python. 

So one can constantly call the `sendMessage` function from multiple different locations and queries will be added to the queue:
https://github.com/openUC2/UC2-REST/blob/ddf92725361fa51bcb082006852c44547a65b098/uc2rest/mserial.py#L340C13-L341C1
They will be send to the ESP one by one. But it seems if we send the json strings too fast they get stuck and there will never be a return (as a qid) which is processed in the other thread with the QID queue. I think this happens if we mix asyn and synced commands - which was the case with first all off, then individual pattern. 

Does this make sense? 
The latest version of the code won't switch off the pattern anymore..