openUC2/ImSwitch #72: Virtual Microscope Improvements

beniroquai, 2024-06-20:
## Modalities
- Light-sheet
- general 3D stuff

## More ideas
- Different Noise models 
- Drift of the stage
- Objective Switching 
- Different Samples 

## Interfaces
- REST API 

**CC:**
@hinderling
@hannasheil
@brunomsaraiva
@jhnnsrs
@henrypinkard
@nornil
@jacopoabramo

beniroquai, 2024-09-10:
@hannahsheil, do you have a stitched stack of a multicolour high-resolution fluorescence sample somewhere by any chance for testing the multi-channel stuff? :)

beniroquai, 2024-09-11:
Great idea with the time-axis ... but let's start with 4d first ^^

beniroquai, 2025-01-08:
A little update: The microscope is life based on a docker container:
https://imswitch.openuc2.com/imswitch/index.html

beniroquai, 2025-08-20:
some inputs:

This simulator would give the option to change the optical resolution (for a green fluorophore) by choosing either of these options:

a 20x/0.75 air objective
a 60x/1.42 oil objective.
The simulator would also give the option to change the digital resolution by choosing either of these options:

changing the pixel size while keeping the same FOV size (this would simulate a point confocal)
adding an extra magnification lens while keeping the same FOV size (this would simulate a camera-based system).
 
This simulator would create aliasing artifacts when undersampling.

The aim is to place students in a situation where they can explore the impact of sampling on the output of image analysis.

Could your simulator do that?

jhnnsrs, 2025-08-21:
not sure if this is of any help: https://github.com/jhnnsrs/faser
is a vectorial psf simulator that also allows to model some params like "coverslip" and SLM settings (phasemask or zernicke polynomials based) in addition to your outlined requirements, but not sure what exactly you are looking for :D

beniroquai, 2025-08-21:
very nice, but already implemented :p