openUC2/ImSwitch #78: Thread in NoQt mode does not perform as intended

beniroquai, 2024-09-05:
Hey @jacopoabramo as I wrote previously, the change you made didn't work out nicely with the current changes I made in order to have something headless-like. I re-introduced you changes in the NoQT Framework

Branch: https://github.com/openUC2/ImSwitch/tree/NOQT

and get the following error that I do not understand:

```py

Defocus:0.0
2024-09-05 22:54:43 ERROR [imcontrol init] Error initializing controller: Can't instantiate abstract class Thread with abstract methods finished, started
ERROR:imswitch:[imcontrol init] Error initializing controller: Can't instantiate abstract class Thread with abstract methods finished, started
2024-09-05 22:54:43 ERROR [main] Failed to initialize module imcontrol
ERROR:imswitch:[main] Failed to initialize module imcontrol
2024-09-05 22:54:43 ERROR [main] Can't instantiate abstract class Thread with abstract methods finished, started
ERROR:imswitch:[main] Can't instantiate abstract class Thread with abstract methods finished, started
2024-09-05 22:54:43 ERROR [main] Traceback (most recent call last):
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/__main__.py", line 149, in main
    view, controller = modulePkg.getMainViewAndController(
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/__init__.py", line 89, in getMainViewAndController
    raise e
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/__init__.py", line 84, in getMainViewAndController
    controller = ImConMainController(options, setupInfo, view, moduleCommChannel)
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/controller/ImConMainController.py", line 32, in __init__
    self.__masterController = MasterController(self.__setupInfo, self.__commChannel,
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/controller/MasterController.py", line 48, in __init__
    self.recordingManager = RecordingManager(self.detectorsManager)
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/model/managers/RecordingManager.py", line 207, in __init__
    self._thread = Thread(target=self.__recordingWorker.run)
TypeError: Can't instantiate abstract class Thread with abstract methods finished, started

ERROR:imswitch:[main] Traceback (most recent call last):
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/__main__.py", line 149, in main
    view, controller = modulePkg.getMainViewAndController(
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/__init__.py", line 89, in getMainViewAndController
    raise e
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/__init__.py", line 84, in getMainViewAndController
    controller = ImConMainController(options, setupInfo, view, moduleCommChannel)
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/controller/ImConMainController.py", line 32, in __init__
    self.__masterController = MasterController(self.__setupInfo, self.__commChannel,
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/controller/MasterController.py", line 48, in __init__
    self.recordingManager = RecordingManager(self.detectorsManager)
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/model/managers/RecordingManager.py", line 207, in __init__
    self._thread = Thread(target=self.__recordingWorker.run)
TypeError: Can't instantiate abstract class Thread with abstract methods finished, started

2024-09-05 22:54:43 INFO [main] init done
INFO:imswitch:[main] init done
Exception ignored in: <function RecordingManager.__del__ at 0x15dc6a5f0>
Traceback (most recent call last):
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/model/managers/RecordingManager.py", line 210, in __del__
    self.endRecording(emitSignal=False, wait=True)
  File "/Users/bene/Dropbox/Dokumente/Promotion/PROJECTS/MicronController/ImSwitch/imswitch/imcontrol/model/managers/RecordingManager.py", line 261, in endRecording
    self._thread.quit()
AttributeError: 'RecordingManager' object has no attribute '_thread'
(imswitch) 
```

Any idea? It works when I remove the abstract method here:
https://github.com/openUC2/ImSwitch/blob/NOQT/imswitch/imcommon/framework/noqt.py#L31

beniroquai, 2024-09-08:
whoops. True. Will check now.

beniroquai, 2024-09-08:
Worked. If you can id appreciate if you can review the use of those threads in the code I merged the Openuc2 master branch now :) 
Thanks!

closed 2024-09-08 by beniroquai

jacopoabramo, 2024-09-08:
I think I already pointed out the necessary adjustments in the PR concerning threading. In case write in the PR which files should be reviewed that I did not review already