openUC2/ImSwitch #85: HIK DLL on Windows not found!

ranranking, 2025-02-21:
@beniroquai 
 

DEBUG [ESP32Manager -> ESP32] Using API version 2
DEBUG:imswitch:[ESP32Manager -> ESP32] Using API version 2
2025-02-21 12:53:43 DEBUG [ESP32Manager -> ESP32] [SendingCommands]:{"task": "/state_get", "qid": 1}

DEBUG:imswitch:[ESP32Manager -> ESP32] [SendingCommands]:{"task": "/state_get", "qid": 1}

{0: [], 1: [], 2: [], 3: [], 4: [], 5: [], 6: [], 7: [], 8: [], 9: []}
/opt/MVS/lib/aarch64/libMvCameraControl.so: cannot open shared object file: No such file or directory
2025-02-21 12:53:44 DEBUG [HikCamManager -> WidefieldCamera] Trying to initialize Hik camera 0
DEBUG:imswitch:[HikCamManager -> WidefieldCamera] Trying to initialize Hik camera 0
2025-02-21 12:53:44 ERROR [HikCamManager -> WidefieldCamera] name 'MvCamCtrldll' is not defined
ERROR:imswitch:[HikCamManager -> WidefieldCamera] name 'MvCamCtrldll' is not defined
2025-02-21 12:53:44 WARNING [HikCamManager -> WidefieldCamera] Failed to initialize CameraHik 0, loading TIS mocker
WARNING:imswitch:[HikCamManager -> WidefieldCamera] Failed to initialize CameraHik 0, loading TIS mocker
2025-02-21 12:53:44 INFO [HikCamManager -> WidefieldCamera] Initialized camera, model: mock
INFO:imswitch:[HikCamManager -> WidefieldCamera] Initialized camera, model: mock
2025-02-21 12:53:44 DEBUG [ESP32LEDLaserManager -> 488 Laser] Laser despeckle enabled
DEBUG:imswitch:[ESP32LEDLaserManager -> 488 Laser] Laser despeckle enabled
2025-02-21 12:53:44 DEBUG [ESP32Manager -> ESP32] [SendingCommands]:{"task": "/laser_act

beniroquai, 2025-02-22:
[Errorlog.txt](https://github.com/user-attachments/files/18923305/Errorlog.txt)

Essentially I have to install a different version for x86 it seems:

```
cd /tmp
tar -xzvf MVS-2.1.2_x86_64_20231116.tar.gz 
cd MVS-2.1.2_x86_64_20231116
tar -xzvf MVS.tar.gz   
bash setup.sh 
```

then I can do 
```
cd /tmp/MVS-2.1.2_x86_64_20231116# cd MVS/Samples/64/Python/
cp GrabImage/GrabImage.py MvImport/
python3 MvImport/GrabImage.py
```

and get 

```
root@3740d95a0f91:/tmp/MVS-2.1.2_x86_64_20231116/MVS/Samples/64/Python# python3 MvImport/GrabImage.py
SDKVersion[0x4010202]
Find 1 devices!

u3v device: [0]
device model name: MV-CS060-10UM-PRO
user serial number: DA1608778
please input the number of the device to connect:0
press a key to stop grabbing.
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[0]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[1]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[2]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[3]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[4]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[5]
get one frame: Width[3072], Height[2048], PixelType[0x1080001], nFrameNum[6]
```

which results in a MVP for HIK:

```dockerfile 
# For multi-architecture builds (requires docker buildx)
# docker buildx create --use
# docker buildx build --platform linux/arm64,linux/amd64 -t docker_hik --load .

# Or, if building on a native system:
# docker build -t docker_hik .
# docker run --privileged -it docker_hik

# syntax=docker/dockerfile:1
#ARG TARGETPLATFORM=linux/arm64
ARG TARGETPLATFORM=linux/amd64
FROM --platform=$TARGETPLATFORM ubuntu:22.04

RUN apt-get update && apt-get install -y \
    wget \
    unzip \
    python3 \
    python3-pip \
    usbutils

RUN mkdir -p /opt/MVS/bin/fonts

RUN cd /tmp && \
        wget https://www.hikrobotics.com/cn2/source/support/software/MVS_STD_GML_V2.1.2_231116.zip && \
        unzip MVS_STD_GML_V2.1.2_231116.zip && \
        if [ "${TARGETPLATFORM}" = "linux/arm64" ]; then \
            dpkg -i MVS-2.1.2_aarch64_20231116.deb; \
        elif [ "${TARGETPLATFORM}" = "linux/amd64" ]; then \
            tar -xzvf MVS-2.1.2_x86_64_20231116.tar.gz && \
            cd MVS-2.1.2_x86_64_20231116 && \
            tar -xzvf MVS.tar.gz && \
            bash setup.sh; \
        fi
    

    
RUN if [ "${TARGETPLATFORM}" = "linux/arm64" ]; then \
    cd /opt/MVS/Samples/aarch64/Python/ && \
    cp GrabImage/GrabImage.py MvImport/GrabImage.py; \
elif [ "${TARGETPLATFORM}" = "linux/amd64" ]; then \
    cd /opt/MVS/Samples/64/Python/ && \
    cp GrabImage/GrabImage.py MvImport/GrabImage.py; \
fi

ENV MVCAM_COMMON_RUNENV=/opt/MVS/lib
ENV LD_LIBRARY_PATH=/opt/MVS/lib/64:/opt/MVS/lib/32:$LD_LIBRARY_PATH

# Copy the entrypoint script into the image
COPY entrypoint.sh /entrypoint.sh
RUN chmod +x /entrypoint.sh

RUN tar -xzvf /tmp/MVS-3.0.1_x86_64_20240902

CMD exec bash
# CMD ["/entrypoint.sh"]
```

beniroquai, 2025-02-22:
@ranranking could you try the last part a s a dockerfile on your computer ? ;-)

closed 2025-05-03 by beniroquai