openUC2/ImSwitch #87: List of Errors and Improvements

beniroquai, 2025-04-23:
- [ ] Stop if memory is full

ethanjli, 2025-08-14:
> After reboot all seetings configs are reset (Forklift @ethanjli )

The reason this is happening for certain kinds of settings is that those settings are stored in a file which is automatically regenerated (from file snippets) during bootup; that file is a temporary file which does not persist across reboots.

If you need an expedient (i.e. quick-and-dirty) workaround, what you can do is copy the contents of the relevant settings file (e.g. `/etc/firewalld/zones/nm-shared.xml`), delete that file (which is actually a symbolic link pointing to the temporary file which contains the actual settings, rather than being a regular file with its own contents storing the actual settings), and then create a new file at the same path with the desired contents. This is effectively a way of persistently opting out from managing those settings through Forklift.

When I initially implemented the system for generating monolithic settings files (e.g. `/etc/firewalld/zones/nm-shared.xml`) from settings file snippets (e.g. everything inside `/etc/firewalld/zones/nm-shared.d/`) in PlanktoScope OS, my original vision was that ad hoc changes made by the user to the monolithic system settings files (e.g. by editing firewall rules via Cockpit) should by default be treated as temporary experiments and thus should not be persisted. This is because it's possible for the user to make settings changes that make the OS inaccessible (e.g. closing off all ports in the firewall, via Cockpit) in a way that would be difficult for the user to manually revert; so far, the only way I've thought of for recovering from such a situation would be to have a way of resetting any user changes upon a reboot, and the simplest implementation of that idea is to avoid persisting those kinds of user settings changes across reboots. So I was thinking that the OS should add (or modify) a settings file snippet (e.g. `/etc/firewalld/zones/nm-shared.d/70-my-custom-app.xml`) containing non-default settings, in order to make such settings changes persist across reboots; and then the GUI would be an interface for that mechanism.

If we instead actually want to have persistence across reboots by default for any changes made by the user to monolithic settings files (e.g. to `/etc/firewalld/zones/nm-shared.xml`, as a result of user actions in Cockpit's firewall settings GUI), I think I can do that*. Then, as soon as the user makes any edits to that settings file, the auto-generated version will be discarded/ignored so that Forklift is no longer be used to manage any of the settings in that file.

* Note to self: in order to do this, instead of symlinking the monolithic settings file in `/etc` to the corresponding autogenerated file in `/run/overlays/generated/etc`, I'd instead just include `/run/overlays/generated/etc` as a lowerdir of the `/etc` overlay.

beniroquai, 2025-08-20:
Sounds good. My "dream" would be to have the ability to set the Wifi via the GUI/Web and then have a fallback (i.e. Ad-hoc ?) option for the case that the SSID is not available. we could also also think about always adding an USB-to-LAN adapter to the microscope as the Opentrons is doing it. 
The way the opentrons lets the user connect is quiet nice: https://support.opentrons.com/s/article/Get-started-Connect-to-your-OT-2-over-Wi-Fi-optional#:~:text=Go%20to%20Devices%20%3E%20Robot%20Settings,Enter%20the%20network's%20password.

they have a desktop client app - which we also have. We could therefore organize files/os-related request via this pathway. 

Let's discuss that soon! 

A related issue: https://github.com/openUC2/TechnicalDocs-openUC2-FRAME/issues/157#issuecomment-3198725239

CC: @gokugiant