openUC2/ImSwitch #172: ImSwitch-OS Onboarding: first-run flow, networking, Tailscale, config & landing page

beniroquai, 2025-09-25:
Together with @ethanjli and @gokugiant we have been discussing the best strategy on how to interact with the microscope the first time. Based on valuable feedback from @Mikrodoktor we have probably identified a rough first milestone for our project: **See an image through the webinterface and change necessary configurations on the way**
Below I formulate the issue and/or necessary steps to reach this goal. It's all very vaguely formulated. The Drawio file for the graph can be found [here](https://raw.githubusercontent.com/openUC2/ImSwitch/refs/heads/master/docs/images/inswitch_onboarding.drawio). 

The documentation could/should live here? https://github.com/openUC2/openUC2.github.io/blob/master/docs/05_ImSwitch/01_Quickstart.mdx

So, we want tomplement the end-to-end onboarding for new customers unpacking a FRAME microscope. Users should reach a landing page via LAN or Wi-Fi, configure networking, Tailscale, ImSwitch, updates, and file presets with safe rollback.

**Diagram**

[image]

## Scope & Owners

* Frontend/Backend (ImSwitch UI + API, config wizard, update UI, presets, firmware trigger) => I guess @gokugiant 
* OS image, networking stack, captive portal, DHCP/DNS, Tailscale, services, persistence, security => I guess @ethanjli 
* Testing and complaining: @Mikrodoktor 

...but we will learn on the way ;-)

I think the Configuration is shared: @gokugiant you probably provide the  schema/UI + API from the ImSwitch/React side and @ethanjli provides storage location, defaults, docker integration, updates through forklift and systemd services.. ?

## User flow 

I can imagine to have the following pathways:

1. **LAN** => discover device => open IP => **Landing Page** (discovery either our installer [imswitchinstaller/main.js at main · openUC2/ImSwitchInstaller ](https://github.com/openUC2/ImSwitchInstaller/) or something like angryipscanner?)
2. **Wi-Fi** => see SSID => connect with PW (youseetoo) => captive portal opens automatically (?) (fallback to IP) => **Landing Page**

Landing Page cards: **Open ImSwitch**, **Network**, **Computer**, **Updates**, **Files**, **Tailscale**. (@ethanjli you have started that already in the pallet repo https://github.com/openUC2/pallet/

## Deliverables

These are some random points that we should discuss! Please complain/change anything :)

### App (ImSwitch) @gokugiant 

* [ ] **Config Wizard** (first-run + re-run)
* [ ] Connect to nearby Wi-Fi (via API to OS) (=> WifiController.py)
* [ ] Restore config from file / Fallback
* [ ] Show current config + validation in the GUI maybe through the ConfigWizard
* [ ] “Safe apply” with auto-rollback on failure - maybe check config for validity/integrity?
* [ ] **ImSwitch settings pages** for:
  * [ ] API/backend settings (host, ports, HTTPS off/on, tokens)
  * [ ] Camera, stage, modules; persist via common config store => I think this would be a dreamm..have a page of any of the components in the microscope and have a dropdown to customize the settings - maybeintegrated via the youseetoo.gihtub.io/configurator at one point :) Some inspriation here https://www.nature.com/articles/s41592-021-01315-z (Actually REACT based)
  
[image]

* [ ] **Files UI** (already existing - mostly for images/videos captured with microscope, no OS stuff - has to be served through docker mounts)
* [ ] ImSwitchConfig rework
  * [ ] “Single source of truth” indicator (which node stores master config); how is it loaded, CLI vs python main.py vs. Docker - where do we store it?
  * [ ] Sideload config via USB drive? 
* [ ] **Error handling**: toast + details, retry, copy diagnostics.
* [ ] **Telemetry/logging**: structured logs for onboarding steps.
* [ ] **Docs**: short first-run guide with screenshots.

###  OS/Networking @ethanjli 
* [ ] **Landing Page (web)** with cards linking to modules below.
* [ ] **Base image** with services (systemd) and health-checks via forklift 
* [ ] **Networking modes**
  * [ ] Wi-Fi STA connect (from API)
  * [ ] AP fallback with captive portal + local DNS (opens landing page)
  * [ ] LAN discovery (mDNS/LLMNR + DHCP reservation hints)
  * [ ] Optional tethering (USB↔Ethernet) and LAN↔Wi-Fi bridge mode
  * [ ] USB to LAN compatibility?
* [ ] **Captive portal**: intercept + redirect to landing; HTTPS-blocking bypass note.
* [ ] **Device discovery**: mDNS `inswitch.local` (and variant) with proper firewall settings
* [ ] **Config store**
  * [ ] Location + format (e.g., `/etc/inswitch/config.yaml` + `/data/…`)
  * [ ] Transactional write with rollback on network loss
* [ ] **Tailscale**
  * [ ] Preinstalled client; first-run auth flow (QR + URL shown in portal)
  * [ ] Status JSON for UI; persistent state; ACL docs
* [ ] **Security**
  * [ ] Firewall defaults (allow LAN portal/API; block WAN where needed)
* [ ] **Updates UI** (maybe through landing page instead?)
  * [ ] Read versions and display in landing page for OS/app/docker/forklift
  * [ ] Trigger app/container updates; progress + logs via landing page?
  * [ ] Firmware updater hook (OS hook?)

...there is probably a lot missing ;) 

## Acceptance criteria

@Mikrodoktor please add your expectation here :)

* [ ] From factory reset, user reaches landing page via **either** LAN or Wi-Fi in ≤2 min.
* [ ] Wizard connects the device to a protected Wi-Fi and **keeps** access (no lock-out).
* [ ] AP fallback works if STA fails; portal opens automatically.
* [ ] Tailscale shows “connected” after auth; status visible in UI.
* [ ] Config changes are atomic; rollback works on failure.
* [ ] Updates show progress and final status; device remains reachable.
* [ ] Support bundle downloadable from UI.
* [ ] Can connect to nearby Wifi 
* [ ] Need to be able to update the device in offline-only mode somehow

ethanjli, 2025-09-26:
> A desktop app (which you have to install)

I think this could be a worthwhile goal to work towards over the next several months, but it adds certain complexities (e.g. related to desktop app signing for macOS & Windows app security stuff, or platform compatibility things, or packaging Linux apps in a way that would work for any distro the customer might want to use, or a number of technical factors which are outside our control and thus harder to troubleshoot) which I'd prefer to avoid in our first or second iteration at simplifying the onboarding process. Let's start with something less ambitious which is imperfect but which would be easy for us 1) to communicate to users and 2) to help users troubleshoot, and then as we do more iterations at improving the onboarding process in the coming months, we can work towards something more sophisticated which might be more approachable for customers who have less experience troubleshooting technology. I'm sure that as we get feedback from our initial customers in the coming months, they'll give us valuable feedback/input which will shape the fundamental design of any desktop app we build, before we have to spend time building that app (which would be time not spent building other things) :)

> Should we rather take USB-C adapters these days?
Maybe you could provide an Ethernet-to-USB-C adapter, and also (optionally?) provide a USB-A-to-USB-C adapter which probably only costs a few euros

ethanjli, 2025-09-26:
For the record, @Mikrodoktor and I had a meeting today to follow up on the discussion here. I took notes for myself about what we discussed: https://www.notion.so/2025-09-26-Armin-Ethan-Discussion-22d4e612c78a8022831dc141ed8771e1?source=copy_link

ranranking, 2025-09-27:
Yes, sure, looks good, I think Alex also ordered before LAN adapter like that.

Maybe on Amazon? Christine wants also some two USB charger, I can order them together